adafruit/Adafruit_TCS34725 #13: Disable function is private.

tgidave, 2016-04-14:
Is there a reason the disable function is private?

Humancell, 2016-09-24:
I just ran into this same issue.  I am using this in a battery powered device, and wanted to use the "sleep" mode ... which seems to be what the disable() function does.

I edited the header, and moved disable(void) up under enable(void) to make it public, and am now calling it for testing ... we'll see if this does what I need ...

But I am also very curious ... why was disable() made private, and not used?

a18rhodes, 2018-11-10:
Same issue here. Seems that this is a very old issue, perhaps Adafruit could update.

a18rhodes, 2018-11-12:
I forked this and updated then submitted a pull request. Hopefully that helps get it fixed!

closed 2018-11-13 by microbuilder