adafruit/Adafruit_TCS34725 #16: Questions on the Code

kaminarihoshi, 2017-03-22:
Hello everybody

I made a post on the adafruit forum concerning the code of ductsoup.

I hope anyone can help me with the last three questions in the post:
https://forums.adafruit.com/viewtopic.php?f=19&t=114034&p=569903#p569903

Thank you for your help.

closed 2017-03-22 by microbuilder

kaminarihoshi, 2017-03-23:
Why did you close my issue?

My questions on a github-code couldn't be answered on the adafruit forums, so I ask them now here.

Strange how people who really want to learn something just get rejected for asking. (Ignoring that the answer could probably also help other people.)

Am I only allowed to write on github when I'm already a professional programmer?

microbuilder, 2017-03-23:
We have support forums for support. Github is really only for pull requests and sending code to fix bugs. None of the paid support staff monitor the Github issues, but the emails go out to a LOT of people so we very actively discourage support happening here for both of those reasons. As for the forum post, support did actually reply. The code in question isn't our own, so it's hard for us to speak about the intentions of the original author. :)