adafruit/Adafruit_TCS34725 #27: gamma correction seems wrong

iZhangHui, 2019-03-07:
The gamma correction calculation formula in [colorview.ino](https://github.com/adafruit/Adafruit_TCS34725/blob/master/examples/colorview/colorview.ino) seems not right.
`for (int i=0; i<256; i++) {

    float x = i;

    x /= 255;

    x = pow(x, 2.5);

    x *= 255;
`

I think it should be x = pow(x, 2.5); --> x = pow(x, 1/2.5);

ladyada, 2019-03-07:
@PaintYourDragon any thoughts?

cedricp, 2019-05-27:
It's an interesting issue, that would need to be studied.

At first, we must know what's the OETF (opto electronic transfer function of a LED), for what I read, brightness is linear with current (here we're driving LED with voltage).
Then we could apply the correct EOTF (electro optical transfer function) and get a correct result.

Note that I'm not using the term "gamma", it's a too much global actually and means nothing.
I'll do some measures and report if you are interested.

cedricp, 2019-05-27:
In an RGB led datasheet (https://www.arduino.cc/documents/datasheets/LEDRGB-L-154A4SURK.pdf), we can see the EOTF. Search for the _forward current vs. forward voltage_ chart.
So the I think a more correct OETF would be something like pow(x, 1/y), where y is still to find.