adafruit/Adafruit_TCS34725 #30: Division by NULL

weinrank, 2019-05-17:
## Intro
We're using an ESP32 equipped with an Adafruit TCS34725 connected via I2C.
Our IDE is platform.io.

Our setup is measuring lux and color temperature.
```
static void tcs_send(void) {
  uint16_t r, g, b, c;
  tcs.getRawData(&r, &g, &b, &c);

  StaticJsonDocument<512> json_doc;
  JsonObject json_data = json_doc.to<JsonObject>();

  json_data["sensor"] = "tcs34725";
  json_data["temp"]   = tcs.calculateColorTemperature_dn40(r, g, b, c);
  json_data["lux"]    = tcs.calculateLux(r, g, b);

  json_send(mqtt_sensor_topic.c_str(), &json_data);
}
```
## Issue
If the room is completely dark, the ESP32 board panics with a  `IntegerDivisionByZero`.
I've not yet measured the exact values causing this issue, but it seems related to the `calculateColorTemperature_dn40` function.

If desired, I'll provide additional information.

ladyada, 2019-05-17:
you'll need to let us know where exactly its happening for us to debug it - please trace it down to the code line :)

tuexen, 2019-05-17:
My guess is that [calculateColorTemperature](https://github.com/adafruit/Adafruit_TCS34725/blob/master/Adafruit_TCS34725.cpp#L371) is called with `r==0`, `g==0`, and `b==0` (corresponding to black) and this results in a problem in [Adafruit_TCS34725.cpp#L387](https://github.com/adafruit/Adafruit_TCS34725/blob/master/Adafruit_TCS34725.cpp#L387), since `X==0`, `Y==0`, and `Z==0`, and therefore `X + Y + Z` is `0`.
@weinrank Can you put in debug output and confirm this?

weinrank, 2019-05-20:
There are two similar problems when r,g,b,c are zero.

https://github.com/adafruit/Adafruit_TCS34725/blob/master/Adafruit_TCS34725.cpp#L387
https://github.com/adafruit/Adafruit_TCS34725/blob/master/Adafruit_TCS34725.cpp#L534

I've added debug output to the function:
```cpp
  gl = 0.136f * (float)r2 + /** Red coefficient. */
       1.000f * (float)g2 + /** Green coefficient. */
       -0.444f * (float)b2; /** Blue coefficient. */

  float lux = gl / cpl;

  char buffer[200];
  snprintf(buffer, sizeof(buffer), "b2 = %d | r2 = %d\n", b2, r2);
  Serial.println(buffer);

  /* A simple method of measuring color temp is to use the ratio of blue */
  /* to red light, taking IR cancellation into account. */
  uint16_t cct = (3810 * (uint32_t)b2) / /** Color temp coefficient. */
                     (uint32_t)r2 +
                 1391; /** Color temp offset. */

  return cct;
```

When covering the sensor with a black scarf, `r2` and `b2` become zero which leads to a division by zero exception.
```
b2 = 1 | r2 = 1
b2 = 0 | r2 = 0

Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d4518  PS      : 0x00060830  A0      : 0x800d27d6  A1      : 0x3ffb1c40  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x3ffc22e4  A8      : 0x800d4515  A9      : 0x3ffb1c20  
A10     : 0x00000012  A11     : 0x00000010  A12     : 0x3f401598  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb 
```

closed 2019-08-06 by weinrank