adafruit/Adafruit_TCS34725 #53: How to get the maximum sampling rate

yanxiang-wang, 2024-10-20:
Although I can set the integration time to `TCS34725_INTEGRATIONTIME_2_4MS`, the number of samples is still very small, far from the theoretical value: `1000/2.4=416` samples.

caternuson, 2024-10-21:
Simply setting the TCS34725's data rate does not guarantee data will actually be collected at that rate. The code that does the reading must also be properly written. For very fast data rates, using interrupts may be needed.

Can you provide short example code that demonstrates the issue?

yanxiang-wang, 2024-10-21:
```
#include <Wire.h>
#include "Adafruit_TCS34725.h"

/* Example code for the Adafruit TCS34725 breakout library */

/* Connect SCL    to analog 5
   Connect SDA    to analog 4
   Connect VDD    to 3.3V DC
   Connect GROUND to common ground */

/* Initialise with default values (int time = 2.4ms, gain = 1x) */
// Adafruit_TCS34725 tcs = Adafruit_TCS34725();

/* Initialise with specific int time and gain values */
Adafruit_TCS34725 tcs = Adafruit_TCS34725(TCS34725_INTEGRATIONTIME_2_4MS, TCS34725_GAIN_1X);

void setup(void) {
  Serial.begin(9600);
  Wire.setClock(400000);
  if (tcs.begin()) {
    Serial.println("Found sensor");
  } else {
    Serial.println("No TCS34725 found ... check your connections");
    while (1);
  }

  // Now we're ready to get readings!
}

void loop(void) {
  uint16_t r, g, b;
  tcs.getRawData(&r, &g, &b, &c);
  Serial.print("R: "); Serial.print(r, DEC); Serial.print(" ");
  Serial.print("G: "); Serial.print(g, DEC); Serial.print(" ");
  Serial.print("B: "); Serial.print(b, DEC); Serial.print(" ");
  Serial.println(" ");

}
```